mudita/MuditaOS #3872: Boot loop crash

GravisZro, 2022-05-26:
# 🐛 Bug report

## 📝 Description

My phone crashes upon boot.  The Mudita logo appears and then grays, indicating it's crashed.  Pressing the power button to boot again results in the same crash happening repeatedly.  I took the battery out for minute and replaced it to see if this was a hardware issue but it continued crashing.  The only way to escape this boot loop crash is to connect it via USB to a computer and then boot it.

The Mudita Center does not detect a crash has occurred when it's run, which may mean that the logging service never started.

I have no idea what caused this or why which makes this bug entirely unreproducible.  I can't seem to find any tools to dump the EEPROM contents or the contents of the internal database to identify what could cause this.

## 🌍 Your operating system and MuditaOS version
1.2.0

## 🤔 Anything else relevant?
Please take my request (#3869) for validating all setting upon being loaded very seriously because it's entirely possible to end up in a very bad state like my phone.  I'm hoping a tool to dump the EEPROM and database are posted but I will be doing a factory reset soon.

pholat, 2022-05-30:
@GravisZro can you please tell:
- (after successful boot) what's the battery level?
- how often does it happen? can you reliably reproduce it now?

pholat, 2022-05-30:
>  I even removed the battery when it was powered off, powered the phone back on and then the behavior persisted.
did you power on the phone without a battery? it won't work this way as USB won't be a reliable source of power here.

Saying that: data without the battery in the phone is not reliable at all. It will just crash randomly when power spike comes. 

Saying that:  did you reproduce it with working battery?

GravisZro, 2022-05-30:
* bad behavior
* power off
* remove battery
* replace battery
* power on
* still bad behavior

pholat, 2022-06-06:
internal reference: might be connected to [this issue](https://appnroll.atlassian.net/browse/MOS-515).

The issue mentioned was a race condition in application initialization which was visible with the onboarding flow.
It will be fixed in the next minor release ( 1.3.0 I believe, but I'm not 100% sure, as there will be a release with just an audio patch and I do not know if it will be named 1.2.1 or 1.3.0 ) 

The only thing I believe you can do now is one of either:
1. reboot the phone a few times, when the bug was reproducing - it managed to start 1 of 10 times - then wait for the next release with the bugfix.
    1.1 if it still does not start - you may do a factory reset and retry, as you are doing the `onboarding` which means that you shouldn't have anything on the phone either way.
2. send the phone back to Mudita with a request to repair or refund.

GravisZro, 2022-06-06:
After performing a factory reset, the bad behaviors have been eliminated.  Unless I'm missing something, this means that bad behavior could be reproduced using sqlite3 setting databases.

I have two suggestions:
1. If the crash reports do not create copies of the database files, they should or at least settings that aren't default.
2. To identify possible configuration issues, the test harness should be run connected to a phone, iterating through every possible configuration.  I don't know if it's possible to replace the settings databases on a consumer phone but if it is, please inform me of how.

I don't believe any further action can be taken without sqlite3 dumps, so I'm closing this issue.

closed 2022-06-06 by GravisZro