mudita/MuditaOS #3889: Mudita OS freezes randomly

grzegorzgrzegorz, 2022-06-01:
# 🐛 Bug report

## 📝 Description

Phone freezes randomly. I cannot see any pattern about this behaviour. The only tip I have is it seems the problem is gone when sim card is removed. I was also using the phone often in places when it was out of range of GSM. So it was often showing it is offline/online - maybe this could be some hint.

Anyway, I moved sim to my old phone and I have now very expensive MuditaTorch on the table. Of course SIM card works perfectly with my old samsung solid.

There are 2 kinds of freezes:
- soft one when clock is working and minutes are refreshed on the screen - it unfreezes after few minutes (the problem occurs regardless of the fact buttons were used or not)
- hard one when clock is also frozen - only battery removal helps in such situation (this one happened during the night and alarm didn't start in the morning)

## 👍 Expected behavior

Phone is responsive 24 hours a day.

## 👎 Current behavior

Phone hangs randomly regardless of the fact it is used or not.

## 🔬 Minimal reproduction

Install SIM card and put the phone on the table. It should freeze after few hours. Try to bring it out of the range of GSM and back.

## 🔍 Provide logs

I am not sure how to provide the logs.

<pre><code>

</code></pre>

## 🌍 Your operating system and MuditaOS version

MuditaOS 1.2.0

## 🤔 Anything else relevant?

I am surprised no on reported this issue here. Many people talk about it in the Mudita forum. I wonder if you did any stability/durability tests. This defect seems to be very obvious one and easily reproducible. 
The phone feels like early Alpha version.

pholat, 2022-06-06:
@grzegorzgrzegorz please report the issue with logs via MuditaCentre. 
As for reproducibility, we managed to reproduce the non-responsive keyboard locally and are working on it. [Internal issue reference](https://appnroll.atlassian.net/browse/MOS-210).

grzegorzgrzegorz, 2022-06-08:
Thanks for feedback @pholat . Is there any chance to actually view this internal issue reference?
I have no rights to use this link now. After logging in I do not see the ticket. Is this private Jira instance?

SP2FET, 2022-07-04:
Hi @grzegorzgrzegorz. Did you report the issue via MuditaCenter?

Lefucjusz, 2022-07-18:
The issue you've described as the soft one should be fixed in the next MuditaOS release.

p-sims, 2022-07-25:
Pure arrived in post last week. There appears to still have various instability issues in MuditaOS. At list one which required battery to be removed. We also had an issue were we wanted to restart (because of signal problem) and it became completely unresponsive in this state:
[image]

Then after maybe 10-20secionds it booted up again:
[image]

When I managed to get it talking to MuditaCenter successfully, I'll try and report some logs.

Lefucjusz, 2022-08-08:
@grzegorzgrzegorz MuditaOS 1.3.0 has been released, please update your phone and if the problem still manifests itself - reopen the issue.

closed 2022-08-08 by Lefucjusz