mudita/MuditaOS #3912: CPU still running while "Depleted Battery" symbol is displayed

GravisZro, 2022-06-10:
# 🐛 Bug report

## 📝 Description

When the battery gets too low, it displays [`Depleted Battery`](https://user-images.githubusercontent.com/670534/173082739-0abb5d3d-0659-48e7-be57-f4546d17fe36.png).  When it is shown, all buttons are disabled but the CPU still processes events.  When a timed alarm occurs after displaying this symbol, it will activate with no way to disable it.

## 👍 Expected behavior

Upon displaying `Depleted Battery`, the device should shutdown entirely.

## 👎 Current behavior

I'm unsure just how active the CPU is but at the very least, it will respond to alarms there were set before `Depleted Battery` was displayed.  Left unattended, this will drain the battery as much as possible.

## 🔬 Minimal reproduction

1. Set an alarm to go off
2. Reduce battery voltage until `Depleted Battery` is displayed.
3. Wait for alarm to expire
4. Alarm will sound.  Buttons will be unresponsive to disable the alarm.  Battery must be removed to disable alarm.

## 🌍 Your operating system and MuditaOS version

1.2.0

## 🤔 Anything else relevant?

I'm presuming there is a hardware cut-off voltage for the battery but to extend the battery life, the phone is disabled before that point.  If the this is the case then this bug could thwart efforts to extend the battery life as it may drain the battery until the hardware limiter kicks in.

kkleczkowski, 2022-06-13:
This is a deeper internal hadware/software issue. You are partialy right - we shouldn't process OS actions when we display this screen, but it's not like we can totally turn off hardware due to the hardware limitations. It has been already noticed and it's reported under MOS-275. Moreover, we'll take a deeper look at the battery handling.

GravisZro, 2022-06-14:
I recognize that electronics like this are never truly off but putting the CPU in the deepest sleep mode comparable to turning the phone "off" would suffice.

Lefucjusz, 2022-06-20:
@GravisZro Unfortunately it's not as easy as it might seem; there are some technical/hardware limitations that we're unable to overcome with the software. As much as we would like to help you, there's little to none (rather none) space for improvements.

As previously mentioned, part of this issue will be fixed in MOS-275.

closed 2022-06-20 by Lefucjusz