mudita/MuditaOS #3914: Inconsistensies in app_bell_greeting_msg string

StoyanDimitrov, 2022-06-11:
`app_bell_greeting_msg` is a key in `English.json`. I see several problems with this string contents:

1. Title case or sentence case - some of the items are all title case, other items are sentence case but most of them use title case only for the first part of the text:
    -  `<text>Good Morning!<br />It's a Beautiful Day!</text>` - example of all title case text,
    - `<text>Rise and shine!<br />Seize the day</text>` - example of all sentence case text,
    - `<text>It's a Beautiful Day<br />Time to rise & shine</text>` - example of probably what is expected.
2. Some of the items are missing the ending punctuation - `<text>Rise and shine!<br />Seize the day</text>`.
3. There is a trailing white-space before the closing tag - `<text>Hello Sunshine!<br />Go brighten the world! </text>`;
4. There is a typo in `<text>Cheers to the day!<br />he future starts NOW!</text>` - there is a missing **T** after the `<br />`.

kkleczkowski, 2022-06-13:
Hi, thank you for noticing it. It will be fixed and it's reported under BH-1527 ticket.

Lefucjusz, 2022-07-25:
@StoyanDimitrov The issue has been fixed and will be present in 1.7.0 release. First and second points of your issue are actually like this by design, so we haven't changed that.