mudita/MuditaOS #3961: Making calls

snan, 2022-07-08:
It often happens that I can't make calls. I dial the number and I can erase but I can't hit the call button. Since I'm in a hurry and really need to make the call quickly I even try to create a contact and maybe call that way (and later delete the contact) but even then it was impossible to call. A couple of minutes later, I could. This is not good for medical situations 💔

SP2FET, 2022-07-11:
Hi @snan,
sorry to hear that. Could you please elaborate on what `can't hit the call button` means? Could you please share with us the phone number format? (eg. 9 numbers)

We have recently rewritten the call handling which will be included in the next release, so hopefully, it shouldn't happen anymore.

Lefucjusz, 2022-08-08:
@snan Call handling has been completely rewritten, so the issue should be fixed now. If the problem persists, please reopen the issue.

closed 2022-08-08 by Lefucjusz