mudita/MuditaOS #3988: Numbers typed during call appearing on the display

p-sims, 2022-07-26:
# 🚀 Feature request

## 📝 Description
If during a call you type numbers (e.g. to choose between menu option or enter codes) these do not currently display on the phone screen.

## 📝 Describe the solution you'd like
It would be helpful if when you type numbers during a call they appear on the screen so a) you can validate you type the right numbers, and b) you can made a note of such information later in the call if needed. Some of have to spend far to much time navigating bank, insurance and other company call centers. These typed numbers could for instant appear on line below the number/name the call is with in smaller font.

## 📝 Describe alternatives you've considered
They could appear at the bottom of the screen above soft key idents

Lefucjusz, 2022-08-01:
@p-sims Thank you for the suggestion, it has been reported to our team under MOS-644 ticket and will be added in one of the future releases.

Lefucjusz, 2024-06-12:
@p-sims this feature was added in MuditaOS 1.10.0, so please update to the latest version (1.12.0) to have it available

closed 2024-06-12 by Lefucjusz