mudita/MuditaOS #3993: Automatic Date and time

p-sims, 2022-07-28:
# 🐛 Bug report

## 📝 Description

Had new Mudita pure just over a week now and had 'automatic date and time' set to ON for most of that (with SIM card in) and it still thinks it's 1970. I've search for other issues and can only find #2360 which seems to just be about whether it's set ON by default. 

## 👍 Expected behaviour

With SIM card in and 'automatic date and time' set to ON is should take hours for time and date to be updated from 1970. Presumable this is done via Network Carrier. 

## 👎 Current behaviour

Never seems to get set, so you have to enter manually, and re enter every time you take battery out.

## 🔬 Minimal reproduction

Turn phone on, with time and date set incorrectly and 'automatic date and time' set to ON. Wait 24h for nothing to happen.

## 🌍 Your operating system and MuditaOS version

v1.2.0

## 🤔 Anything else relevant?
How long would you expect this to take to get time and date from Carrier?

Lefucjusz, 2022-08-01:
@p-sims Thanks for reporting the issue. Automatic time and date synchronization is highly dependent on your location (precisely: whether your network operator implements such feature in their infrastructure), in some locations it might not work because of this.

esope, 2022-08-02:
Automatic time and date is not working for me either.

Lefucjusz, 2022-10-10:
@p-sims The fundamental idea of Mudita Pure is to not provide any Internet connection. Thus it is impossible to make any HTTP requests to fetch any data. The only way for time setting is automatic synchronization provided by the GSM network or just setting it manually.

closed 2023-04-03 by Lefucjusz