mudita/MuditaOS #3996: Sim card

snan, 2022-08-01:
I think a lot of the issues— can't call, can't text, drops calls, drops texts, hangs up mid call, missed calls—are because it sometimes believes that there's no sim card in there 🤷🏻‍♀️

snan, 2022-08-01:
Usually restarting it is what makes it understand that it's there

Lefucjusz, 2022-08-05:
@snan Thanks for reporting the problem, we're sorry for your inconvenience. Please send us logs via Mudita Center, so we can investigate this issue further. 
Quick description how to do this can be found here: https://forum.mudita.com/t/how-to-send-logs-to-mudita-via-mudita-center/4085

Lefucjusz, 2022-08-08:
@snan Please also provide the serial number of your phone here after you've sent the logs.

Lefucjusz, 2022-10-10:
Hi @snan, could you provide us with the logs and detailed description of the issue you've been experiencing?

Lefucjusz, 2023-01-02:
Closing the issue due to lack of response.

closed 2023-01-02 by Lefucjusz