mudita/MuditaOS #3997: Unread messages (but I can't ever read them because they're deleted)

snan, 2022-08-02:
I got a few messages by a harasser and deleted the thread and now my homescreen perptually says unread messages and I can't decrement that count.

Lefucjusz, 2022-08-05:
Hi @snan, thanks for reporting the issue! We are working on improving the automatic refreshing of notifications in subsequent releases under MOS-578 and MOS-651 tickets. For now you should be able to remove this notification by focusing on it and clicking CLEAR.

Lefucjusz, 2023-06-12:
Hi @snan, the issue has been resolved and will be released in one of the next MuditaOS releases. I'm closing this ticket now, but feel free to reopen it if you encounter such problem in the future.

closed 2023-06-12 by Lefucjusz