mudita/MuditaOS #4011: calling operator codes

dr3mro, 2022-08-07:
dialing operator codes #999# don't request the code but it dials number 
dialing *99# code will return long line of random numbers like 019298201980217987623962876387263872863
will this issue be fixed ?

Lefucjusz, 2022-08-22:
@dr3mro Thanks for reporting the problem. You've also sent your request to our Customer Support, so we will keep you updated there as we work on this issue.

dr3mro, 2022-10-17:
Any idea when this issue will be fixed?

Lefucjusz, 2022-10-17:
@dr3mro At the moment different issues have highest priority on our roadmap, but we'll move the priority of this matter.

dr3mro, 2022-10-17:
[image]

[image]

Please as its it required to charge my phone with credit and control my line feature for example dont disturb mode from the operator itself

Lefucjusz, 2023-01-09:
Hi @dr3mro! We've added additional log in newest MuditaOS release - 1.5.0 - so that we can gather more information regarding your issue. Please update to 1.5.0, send the code that has been causing your problem and after that send us the logs via Mudita Center, so we can investigate further. Detailed instruction on how to send the logs can be found here: https://forum.mudita.com/t/how-to-send-logs-to-mudita-via-mudita-center/4085.

Lefucjusz, 2023-01-16:
Hi @dr3mro, have you managed to update your phone and follow the procedure I described?

dr3mro, 2023-01-16:
ok I just sent you the logs and here is a video while I dial the codes

https://drive.google.com/file/d/1MlZYawL9_qKqVr9uAAjngHHE160cpTer/view?usp=sharing

Thank You
Dr. Amr Osman

dr3mro, 2023-03-24:
after updating to 1.6 and now 1.7 dialing *number# codes does nothing at all and #number# still doing calls even if I found a git commit that should have fixed that

Lefucjusz, 2023-03-24:
Hi @dr3mro! Unfortunately the change implemented in [commit you're probably talking about](https://github.com/mudita/MuditaOS/commit/5564408a56280fa09eada40c038e1c7ee48b8287,) has not been released as a part of MuditaOS 1.7.0 - it will be added to the scope of next OS release.