mudita/MuditaOS #4015: Automatic Send SMS to keep SIM card alive

p-sims, 2022-08-10:
# 🚀 Feature request

## 📝 Description

[Most SIMs get deactivated if you don't send an text or make a call within between 70 days and 6months (at last in the UK)](https://www.ofcom.org.uk/phones-telecoms-and-internet/advice-for-consumers/advice/pay-as-you-go-mobile-use-it-or-lose-it)

Could there be a really simple automatic feature in settings where you could enable an automatic send a text every 1/3/6 months to a particular number with particular message. That message could by default be "This is an automatic text send to stop SIM card being deactivated.".

This would just be a really nice feature for people who want a simple phone they don't use much.

## 📝 Describe the solution you'd like

New menu under 'Network' called "SIM Keep Active" with the following sub items:
- SIM1 Keep Active [ON/OFF]
- SIM2 Keep Active [ON/OFF]
- SMS Destination (For setting number to send text to)
- SMS Content (Default: "This is an automatic text send to stop SIM card being deactivated.")

Ideally (like with Alarms) when enable this should set a watch dog timers so phone will boot up even if turned off, send the SMS and then shutdown again.

## 📝 Describe alternatives you've considered

Could make call not send SMS and then hang up at point connected but that would be more confusing for recipient.

lennie420, 2022-08-15:
What's the use of having a phone if you don't use it this often? I think it's a bit unfair to expect the developers to program such a feature.

p-sims, 2022-08-15:
I'm not saying it should be top priority new feature.

p-sims, 2022-08-15:
Sorry I didn't mean to close the thread.

I'm not saying it should be top priority new feature, it might also be something I was willing to contribute to implementing rather than relying on others to implement.

> What's the use of having a phone if you don't use it this often?

I know many elderly people who want a simple phone for emergency use (e.g. break down, or walling while out and about). I also know people including myself who would like to have a simple phone (with it's own sim and dedicate number that they use on holiday, on protests, or when they want to be away from the notification barrage of their work phones. It is all too easy for sim's in such phones to have become deactivate because you didn't send a text or make a call within 61 days, which defeats the point of an 'emergency phone'. I also know people who prefer not to have a phone at all, but have one so they can make emergancy calls, or recieve texts from others.

Lefucjusz, 2022-08-22:
@p-sims Thank you for taking the time to send us your suggestions. Currently we are working on stabilizing everything in Mudita Pure so that it functions well for users.

Lefucjusz, 2022-10-10:
@p-sims After a discussion with the rest of the team we came to the conclusion that this is very custom functionality and rather will not be implemented.

closed 2022-10-10 by Lefucjusz