mudita/MuditaOS #4055: Phone Ringtone in Music Volume, rather than Volume

LizGalanis, 2022-09-06:
# 🐛 Bug report
It seems that the phone call ringtone volume is the same as the 'music' volume, while alarm and SMS ringtone volume are simply controlled with 'volume' from other menus. 

## 📝 Description
For several days I did not have any ringtone when receiving calls. I thought something was going wrong with the phone, until I realised when receiving one call the music was very soft, despite the volume being turned up high. I went to the phone app to change the ringtone. When I tried to play ringtones they were they same softness as with incoming calls. I turned up the volume and the volume bar with 'music' as the title appeared. I believe phone ringtone is controlled in sync with the 'music' volume while everything else is controlled by another volume. This means, if I turn down my music in my headphones one time, my incoming call ringtone is too quiet and out of sync with SMS ringtone. 

A clear and concise description of the problem...

## 👍 Expected behavior

What you think should happen...

I think that incoming calls, SMS and alarm should be controlled through one volume, seperate to music volume

## 👎 Current behavior

Current 'buggy' behavior...

Incoming calls and music controlled with the same volume, SMS and alarm with another volume

## 🔬 Minimal reproduction

Simple steps to reproduce this bug.

Change the volume while listening to music and you will see this changes the incoming call volume

Issues that don't have enough info and can't be reproduced will be closed.

## 🔍 Provide logs

Please paste relevant logs below or attach them to this issue.

<pre><code>

</code></pre>

## 🌍 Your operating system and MuditaOS version

## 🤔 Anything else relevant?

Lefucjusz, 2022-09-12:
@LizGalanis thanks for reporting the issue, we'll investigate if further and update you as soon as we gather some more knowledge. Can you please specify:

- what type of headphones you used when the problem happened? Were they wireless Bluetooth or wired? 
- Did you change the volume from headphones' buttons or using Pure's buttons?
- Have you noticed the issue when using phone without headphones?

Lefucjusz, 2022-11-14:
@LizGalanis I'm closing this issue now due to lack of response.

closed 2022-11-14 by Lefucjusz