mudita/MuditaOS #4056: Dates on SMSs

LizGalanis, 2022-09-06:
# 🚀 Feature request
At the moment I can only see the time that an SMS was sent/received, cannot see the date even if it was sent weeks ago. I find this impractical when going through previous messages. 

## 📝 Description

A clear and concise description of the problem or missing capability

I needed to check on what had happened on a particular date at work, and I new that I could check through my SMSs. However, when I scrolled through realised I had no idea which messages were from what date. Caused me problems. 

## 📝 Describe the solution you'd like

If you have a solution in mind, please describe it.

Would like to see the dates on SMSs, in particular if they were not sent today

## 📝 Describe alternatives you've considered

Have you considered any alternative solutions or workarounds?

Lefucjusz, 2022-09-12:
@LizGalanis thanks for the suggestion, this feature has already been requested, its internal task number is MOS-653 - it will be added in one of the further releases.

Lefucjusz, 2023-03-27:
Hi @LizGalanis - the issue has been fixed and is already released in MuditaOS 1.7.0.
After the change:

* for the today's messages only hour is shown;
* for the yesterday's message "yesterday" and hour is shown;
* for the messages from the last week day of the week ("Mon"-"Sun") and hour is shown;
* for the older messages short form of date is shown ("DD.MM") together with hour;
* for the messages from the previous years full date is shown ("DD.MM.YY") together with hour.

closed 2023-03-27 by Lefucjusz