mudita/MuditaOS #4057: Phone Constantly Pocket Dialing

LizGalanis, 2022-09-06:
# ⚠️ Create other issue

## 📝 Description

A clear and concise description of the problem or missing capability...

My phone is constantly pressing numbers in the pocket in lock mode and trying to make an emergency call. I think this impacts battery life, however I'm also worried that one day I could really dial an emergency number and make an accidental call. Another issue is that when I open up the phone, I have to delete individually every number that has been pressed in the pocket, could be up to 20 numbers. 

## 📝 Describe the solution you'd like

If you have a solution in mind, please describe it.

I would prefer the phone not to be able to have any functions/make calls in lock mode. 

## 📝 Describe alternatives you've considered

Have you considered any alternative solutions or workarounds?

Alternatively, at the very minimum, I would appreciate being able to delete all the numbers in the emergency call at once, rather than one by one. However, ultimately, would not like the phone to do anything while in lock.

LizGalanis, 2022-09-06:
Issue seems to be different now. Unsure what happened

Lefucjusz, 2022-09-12:
@LizGalanis thanks for reporting the issue. However, I'm afraid there's not much that can be done in this matter - every phone has to have a capability to make an emergency call without the need to unlock it before, so it's possible that it can be mistakenly dialed. As for the need to delete every digit individually - you can just hold "Clear" button and whole number will be deleted at once.

closed 2022-09-15 by Lefucjusz