mudita/MuditaOS #4061: #100# code dial instead of send as code

dr3mro, 2022-09-09:
trying to request #100# code to tweak some network services on Orange Egypt actually try to dial instead of send it as code
please fix that

Lefucjusz, 2022-09-12:
@dr3mro thanks for reporting the issue. However, as far as I understand, you've already reported this bug here: https://github.com/mudita/MuditaOS/issues/4011

We created the ticket for this problem already, MOS-655, and will be working on it in the future - for now we have tasks with higher priority on our roadmap.

Lefucjusz, 2023-03-27:
Hi @dr3mro - as you already noticed, the issue has been fixed and will be released in one of the next versions of MuditaOS. We'll keep you informed in this thread: https://github.com/mudita/MuditaOS/issues/4011 - I'm closing this one now.

closed 2023-03-27 by Lefucjusz