mudita/MuditaOS #4125: Answered calls are marked as "unseen"

esope, 2022-10-25:
# 🐛 Bug report

## 📝 Description

When you successfully answer a call, this call is marked as "unread" in the list of calls, i.e.: there is a fat dot on the icon for calls in the main menu, and the call itself is displayed in bold.

## 👍 Expected behavior

When I answer a call, the call should be listed in the list of calls, but it should be displayed as an "already seen" call (not in bold, and with no fat dot on the calls icon in the main menu).

## 👎 Current behavior

When I answer a call, the call should is listed in the list of calls, but it is displayed as an "unseen" call (in bold, and with a fat dot on the calls icon in the main menu).

## 🔬 Minimal reproduction

Answer a call. Terminate the conversation. Open the main menu: there is a fat dot on the calls icon. Enter the list of calls: the call you've just answered is marked in bold.

## 🔍 Provide logs

Ask me if needed.

## 🌍 Your operating system and MuditaOS version

MuditaOS 1.4.0

## 🤔 Anything else relevant?

Nope.

Lefucjusz, 2022-12-19:
Hi @esope, thanks for reporting the issue. We'll check whether such issue can be reproduced and let you know as soon as we know more in this matter.

Lefucjusz, 2023-01-09:
Hi @esope, after internal testing we confirmed that the bug exists indeed, however we were able to reproduce only the "fat dot" issue - the text is not in bold. The issue has been reported and will be fixed in one of the further releases - internal ticket's code: MOS-870.

Lefucjusz, 2023-02-09:
Hi @esope! The bug has been resolved recently and the fix will be present in one of the next MuditaOS releases.

closed 2023-02-09 by Lefucjusz