mudita/MuditaOS #4149: Pure alarm timeout apparently not affected by playDurationTimer

sharethewisdom, 2022-11-19:
# 🐛 Bug report

## 📝 Description

The alarm apparently plays back indefinitely by default. There is no option for me to specify a timeout in the alarm GUI settings. 

## 👍 Expected behavior

I think most users would appreciate a generous default timeout of half an hour or so, certainly not indefinite. If I understand the code correctly, the `playDurationTimer` timer in `PlayAudioActions.cpp` from the PurePhone alarm actions should disable the playback. So there may be an issue in the code.

## 👎 Current behavior

I forgot the phone at work a few times, and I noticed that the alarm was still active after three hours. Two observations:

- At one time, the alarm audio was playing and the vibration was active, but I did not see a popup to disable the alarm, and I could not go to a screen to do so. The only solution I found was to turn the phone off and on again.
- Another time, alarm audio was playing and the vibration was active, but the screen showed the dead battery icon (i.e. the phone was "off"). The only solution to disable the alarm I found was to remove the battery.

## 🔬 Minimal reproduction

- set up an alarm
- observe that it plays back indefinitely
- observe that there are GUI issues then (described above)

## 🌍 MuditaOS version

MuditaOS 1.4.0

Lefucjusz, 2022-11-28:
Hi @sharethewisdom!

Thanks for letting us know about the problem. The issue has been reported under MOS-829 ticket and will be fixed in one of the further releases.

Lefucjusz, 2023-01-23:
Hi @sharethewisdom, the issue has been resolved - we introduced a timeout, which will automatically disable the alarm after 30 minutes. The fix will be present in one of the next MuditaOS releases.

Lefucjusz, 2023-01-30:
I'm closing this issue now, if you'd like to know in which version the fix will appear, please take a look at the changelog in MuditaOS repo.

closed 2023-01-30 by Lefucjusz