mudita/MuditaOS #4199: Hard to Call People Not in Contacts

sarahspock, 2022-12-08:
# 🚀 Feature request
It is not apparent in the UI on how to call a number (example: customer service hotline) that is not in the contacts. It would be nicer if how to do this is more obvious without adding the number to the contacts

## 📝 Description

I want to be able to call numbers easily without adding to contacts or receiving a text or message from the number. An example use case is needing to call a customer service hotline.

## 📝 Describe the solution you'd like

Open calls app and press left arrow to open call prompt that shows numbers entered into phone. Allow numbers to be pressed for the call

## 📝 Describe alternatives you've considered

I just add customer service lines to contacts before calling them

Lefucjusz, 2022-12-12:
Hi @sarahspock, thanks for reporting the problem. I'm not sure whether I've understood it correctly - you tried to enter some number in Contacts app screen and you weren't able to do so? If that's the case - such functionality is possible on the desktop - just unlock the phone and enter the number using the keypad. If that is not the case - please clarify your issue.

sarahspock, 2022-12-12:
This is just a UI feature request.  No, try calling a number from customer service in the UI , for example not in the contacts. You will find how to do so impossible or very hard to find. For example, going to calls you cannot make a call from a non-contact. I think an obvious feature for the phone should be being able to call people that aren't listed in contacts.

Lefucjusz, 2023-01-16:
Hi @sarahspock, can you please let us know how you are imputing the number which you are trying to dial?  Are you putting in +1(AREA CODE) and then the number?

Lefucjusz, 2023-03-27:
Closing due to lack of activity.

closed 2023-03-27 by Lefucjusz