mudita/MuditaOS #4200: Phone loses signal when switching phone towers and does not receive messages / calls

sarahspock, 2022-12-08:
# 🐛 Bug report
Switching cell phone towers causes phone to not receive messages and calls due to no signal

## 📝 Description

When traveling from a location A to a location B , if you switch phone towers, the phone receives a signal and then does not receive a signal when it switches towers. You haver to toggle the kill switch on the phone from connected to offline and then to connected again when switching phone towers in order to receive a signal and calls. This was tested on the tmobile network in the USA

## 👍 Expected behavior

The phone should at the very least perform the toggle functionality automatically or ideally perform seamless cell phone tower handoff. People may forget to manually toggle the kill switch and miss calls and messages

Lefucjusz, 2022-12-12:
Hi @sarahspock, thanks for reporting the issue. To perform further investigation of this problem we need to know one additional thing - does the signal returns back after some time if it has been lost?

zabbal, 2022-12-12:
On a related note: you can replicate this situation in your lab using http://osmocom.org

sarahspock, 2022-12-12:
The signal does not return.

dr3mro, 2022-12-12:
same issue with me since I first used it in Egypt, and tried change operators and is the same

Lefucjusz, 2022-12-19:
Hi @sarahspock. To investigate your issue further, we'll need to analyse the logs from the phone. Please send us the logs via Mudita Center, the instruction how to do it can be found here: https://forum.mudita.com/t/how-to-send-logs-to-mudita-via-mudita-center/4085

sarahspock, 2022-12-22:
I sent the logs, let me know if you did not receive it

Lefucjusz, 2023-01-16:
Hi @sarahspock, sorry for the late response. Please provide us with your phone's serial number here, so we can find the related logs and continue to analyze your issue.

dr3mro, 2023-01-16:
27880710051922

Grimdur82, 2023-08-22:
Is there anything new about the problem?
I have exactly the same problem. When I drive from one point in the city to another, I no longer receive messages and calls afterwards. I have to remove the battery first and then everything works again. Unfortunately, the phone is not really usable.