mudita/MuditaOS #4278: Mudita mobile TOTP application

pjgg, 2023-01-22:
# 🚀 Feature request

## 📝 Description

As a Mudita mobile user, I would like to have a TOTP application (Time-based one-time password) in order to be able to login into several services such as Github o my personal email. 

## 📝 Describe the solution you'd like

a TOTP app is just a simple application that generates a one-time password based on a secret seed. Normally, you use to scan a barcode in order to get the secret, but is also possible to introduce de secret by yourself as Text. 

So imagine that your Github secret is: 
```
Your two-factor secret: WWWE2N5JKKK1OAB9
```

The mudita TOTP app should request an appName and the secret, and then be able to generate a valid TOTP. 

## 📝 Describe alternatives you've considered

Impl example: https://github.com/RavuAlHemio/cpptotp/blob/master/src/gauche.cpp#L4

Lefucjusz, 2023-01-30:
Hi @pjgg, thanks for your suggestion! There's already been some idea proposed by one of the GitHub community users in this issue https://github.com/mudita/MuditaOS/issues/1823 - unfortunately the problem that's described there still exists - we're unable to store the seed for such password generator safely, as Pure doesn't have any encrypted storage implemented.

closed 2023-01-30 by Lefucjusz

pjgg, 2023-01-30:
Woo, I see... and what about if the app requests you an encryption password at the beginning a "Main password", that could be used to encrypt the seeds through a symmetric algorithm like AES and this password is not stored must be provided by the end user when you open the app (as other security apps are doing like [bitwarden](https://bitwarden.com/))