mudita/MuditaOS #4389: Mudita Pure becomes unresponsive after a few days

esope, 2023-03-06:
# 🐛 Bug report

## 📝 Description

After several days of uptime and regular use (SMS and phone calls), the Mudita Pure becomes unresponsive : the keyboard does not work anymore, but the Offline/DND/Online switch seems to work.
The only way to get back a normal behavior is to remove the battery and reboot the phone.

## 👍 Expected behavior

The Pure should remain responsive all the time.

## 👎 Current behavior

The Pure becomes unusable after a while.

## 🔬 Minimal reproduction

Several days of regular use of the phone without reboot.

## 🔍 Provide logs

## 🌍 Your operating system and MuditaOS version

MuditaOS 1.6.0

## 🤔 Anything else relevant?

This issue was not happening with the previous version of MuditaOS.

Lefucjusz, 2023-03-06:
Hi @esope, thanks for reporting the issue. Please send us logs via Mudita Center so we can investigate your problem further, the instruction on how to do it can be found here: https://forum.mudita.com/t/how-to-send-logs-to-mudita-via-mudita-center/4085

esope, 2023-04-12:
Done!

closed 2023-12-23 by esope