mudita/MuditaOS #4421: Signal Messenger

nicholascioli, 2023-03-17:
# 🚀 Feature request

## 📝 Description

I know that the point of a device like this is to minimize the dependency on Apps and simplify the experience, but even other minimalist devices (like the Punkt MP02) support running Signal messenger as it is used for secure messaging. I'd love to see this implemented, as SMS messages are inherently insecure.

## 📝 Describe the solution you'd like

As MuditaOS is based off of FreeRTOS, this should be fairly straightforward to add in terms of TCP/IP support, as FreeRTOS has [libraries available](https://www.freertos.org/http/index.html) for adding networking. As for signal support, the rust library could be used.

## 📝 Describe alternatives you've considered

Alternatives would include either setting up a Signal <> SMS bridge on a separate device, but then that would defeat the entire point of secure messaging with Signal.

Lefucjusz, 2023-03-20:
Hi @nicholascioli, thanks for reporting the request. Unfortunately, in this Pure's iteration we're unable to implement Signal app as a part of our OS - this has already been verified.

joshspicer, 2024-01-16:
@Lefucjusz does this mean the hardware in incompatible (ie, anyone that purchased a phone can expect to never have a Signal app in an update?)