mudita/MuditaOS #4436: UI inconsistency

dr3mro, 2023-03-30:
After last update the timeout is inconsistent between volume and phone modes 
if you change the volume it takes 3 seconds for it to disappear but changing mode only takes 1 second
please make all take 1 second
thanks
amr

Lefucjusz, 2023-04-03:
Hi @dr3mro, thanks for your suggestion. We'll discuss it with our design team and come back to you with the answer.

Lefucjusz, 2023-04-11:
Hi @dr3mro, as I promised - I'm coming back with the answer. I've talked to our design team and their recommendation is to change the volume popup's display time to match phone modes popup. I've seen you'd created a PR for this change, but then you closed it. If you'd like to contribute to MuditaOS, please create new PR with the change, following our contribution guideline regarding commit message title: https://github.com/mudita/MuditaOS/blob/master/doc/development_workflow.md#commit-changes. I've created the ticket on our Jira for you, the issue number is MOS-980.

Lefucjusz, 2023-04-27:
Closing due to lack of response. The issue has been fixed and will be released in one of next MuditaOS versions.

closed 2023-04-27 by Lefucjusz

dr3mro, 2023-04-27:
That's great 🥳