mudita/MuditaOS #4468: Error while trying to build MuditaOS

olegKusov, 2023-05-01:
#  📚 Missing or unclear documentation

## 📝 Description
What im doing:
1. added github api token with help of this guide (https://github.com/mudita/MuditaOS/blob/master/doc/download_assets.md)
getting:
```
oleg@DESKTOP-713Q8QS:~/projects/MuditaOS/build-PurePhone-linux-Debug$ curl -o /dev/null -H "Authorization: token $(git config user.apitoken)" "https://api.github.com/repos/mudita/PureUpdater"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   117  100   117    0     0    180      0 --:--:-- --:--:-- --:--:--   180
oleg@DESKTOP-713Q8QS:~/projects/MuditaOS/build-PurePhone-linux-Debug$

```

and 

```
oleg@DESKTOP-713Q8QS:~/projects/MuditaOS$ python3 ./tools/download_asset.py github --repository PureUpdater list
download_assets.py
Traceback (most recent call last):
  File "/home/oleg/projects/MuditaOS/./tools/download_asset.py", line 582, in <module>
    downloader.list_releases()
  File "/home/oleg/projects/MuditaOS/./tools/download_asset.py", line 508, in list_releases
    from tabulate import tabulate
ModuleNotFoundError: No module named 'tabulate'
oleg@DESKTOP-713Q8QS:~/projects/MuditaOS$

```
I don't try to investigate on error above. And then step 3:

3. running  ./configure.sh pure linux debug
2. then run "cd build-PurePhone-linux-Debug && make -j 2 Pure"

Why make with -j 2 and not ninja? Because I have laptop with 8 gb of RAM with WSL and I'm getting error related to not enough memory.

With command on step 2 I'm getting error

```
[  2%] Building CXX object third-party/hash-library/CMakeFiles/hash-library.dir/src/sha1.cpp.o
[  2%] Building CXX object third-party/hash-library/CMakeFiles/hash-library.dir/src/sha256.cpp.o
[  2%] Building CXX object third-party/hash-library/CMakeFiles/hash-library.dir/src/sha3.cpp.o
Upgrading 'settings_v2', devel features: False
-> Upgrading to committed version: 0
[  2%] Linking CXX static library ../../lib/libhash-library.a
[  2%] Built target hash-library
[  2%] Building CXX object third-party/json/CMakeFiles/json.dir/json11.cpp.o
-> Upgrading to the newest available revision: 06e96a01-c32c-49b7-8909-43b310f181ff
 -> Running upgrade from 0 to 06e96a01-c32c-49b7-8909-43b310f181ff
[  2%] Linking CXX static library ../../lib/libjson.a
Upgrading 'predefined_quotes', devel features: False
-> Upgrading to committed version: 0
[  2%] Built target json
[  2%] Generating icalderivedproperty.c
[  2%] Generating icalderivedparameter.c
[  2%] Generating icalrestriction.c
  0%|                                                                                               | 0/16 [00:00<?, ?it/s][  2%] Generating icalderivedvalue.c
[  2%] Generating icalderivedproperty.h
Traceback (most recent call last):
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 426, in download_json
    val = Verse(**val, sha_gen=lambda _: self.fallback_ref())
  File "<string>", line 10, in __init__
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 138, in __post_init__
    self.ref = self.sha_gen(self)
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 426, in <lambda>
    val = Verse(**val, sha_gen=lambda _: self.fallback_ref())
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 413, in fallback_ref
    for val in self._api_list_branches():
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 400, in _api_list_branches
    return self.api.repos.list_branches()
  File "/home/oleg/.local/lib/python3.10/site-packages/ghapi/core.py", line 61, in __call__
    return self.client(self.path, self.verb, headers=headers, route=route_p, query=query_p, data=data_p)
  File "/home/oleg/.local/lib/python3.10/site-packages/ghapi/core.py", line 120, in __call__
    res,self.recv_hdrs = urlsend(path, verb, headers=headers or None, debug=debug, return_headers=True,
  File "/home/oleg/.local/lib/python3.10/site-packages/fastcore/net.py", line 218, in urlsend
    return urlread(req, return_json=return_json, return_headers=return_headers)
  File "/home/oleg/.local/lib/python3.10/site-packages/fastcore/net.py", line 119, in urlread
    if 400 <= e.code < 500: raise ExceptionsHTTP[e.code](e.url, e.hdrs, e.fp, msg=e.msg) from None
fastcore.net.HTTP404NotFoundError: HTTP Error 404: Not Found
====Error Body====
{
  "message": "Not Found",
  "documentation_url": "https://docs.github.com/rest/branches/branches#list-branches"
}

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 578, in <module>
    downloader.download_json()
  File "/home/oleg/projects/MuditaOS/tools/download_asset.py", line 440, in download_json
    raise RuntimeError(f'file not found with: {data} err: {ex}')
UnboundLocalError: local variable 'data' referenced before assignment
[  2%] Generating icalderivedparameter.h
[  4%] Generating icalderivedvalue.h
make[3]: *** [products/PurePhone/CMakeFiles/json-proprietary-target.dir/build.make:71: products/PurePhone/CMakeFiles/json-proprietary-target] Error 1
make[2]: *** [CMakeFiles/Makefile2:14172: products/PurePhone/CMakeFiles/json-proprietary-target.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  4%] Generating ical.h
[  4%] Built target ical-header
Trying to download release from MuditaOSPublicAssets under name: MuditaOSPublicAssets 0.0.9-pure
[  4%] Built target json-common-target
Upgrading 'multimedia', devel features: False
-> Upgrading to committed version: 0
Upgrading 'notes', devel features: False
-> Upgrading to committed version: 0
Upgrading 'custom_quotes', devel features: False
-> Upgrading to committed version: 0
Upgrading 'notifications', devel features: False
-> Upgrading to committed version: 0
Upgrading 'calllog', devel features: False
-> Upgrading to committed version: 0
Upgrading 'events', devel features: False
-> Upgrading to committed version: 0
Upgrading 'sms', devel features: False
-> Upgrading to committed version: 0
-> Upgrading to the newest available revision: 111a2d1b-dc32-40a3-9a2a-02a9e186bcac
 -> Running upgrade from 0 to 111a2d1b-dc32-40a3-9a2a-02a9e186bcac
[  4%] Built target create_product_databases
make[1]: *** [CMakeFiles/Makefile2:14210: products/PurePhone/CMakeFiles/Pure.dir/rule] Error 2
make: *** [Makefile:3780: Pure] Error 2
```

A clear and concise description of the problem or missing documentation.

## 📝 Describe the solution you'd like

If you have a solution in mind, please describe it.

## 🔗 Link to an article that could be improved

Please support us with the link to the current article that could be improved.

Lefucjusz, 2023-05-08:
@olegKusov Hi, thanks for reporting the issue! 

**As for the`ModuleNotFoundError: No module named 'tabulate'`** - you need to install `tabulate` module for Python as it is missing in your environment, that should fix the issue. To install all the required dependencies please execute the commands:

```
python3 -m venv venv
source venv/bin/activate
pip3 install -r docker/assets/requirements.txt
```

Run them from the root of MuditaOS directory.

**As for the second issue** - it is most likely caused by the missing CMake flag required for community builds. Please see `https://github.com/mudita/MuditaOS/blob/master/doc/quickstart.md#project-configuration`, there's an info about this: 

```
To build MuditaOS with no access to proprietary files, please specify cmake flag: -DASSETS_TYPE=Community
```

Please try to reconfigure the project with that additional flag and let me know whether it helped.

Lefucjusz, 2024-08-06:
Closing due to lack of discussion activity.

closed 2024-08-06 by Lefucjusz