mudita/MuditaOS #4563: Call waiting

dr3mro, 2023-08-02:
I am using mudita pure as  my daily driver now and I have noticed that whenever I am on a call and someone tries to call me it never shows missed call like any other phone 
Expected behaviours is to show a missed call when call waiting is active 
I do hear the waiting tone but no indication about who called me 
I wish I did explain it well
Thanks

Lefucjusz, 2023-08-07:
Hi @dr3mro, we have received reports about this topic already, the issue is in our backlog under https://appnroll.atlassian.net/browse/MOS-817 ticket. We'll inform you when it is resolved.

dr3mro, 2024-05-28:
any news about this issue as it causes a lot of problems as some one calls when we are in  the middle of another call