mudita/MuditaOS #4577: Navigation support GoogleMaps or something like that

alok-dks, 2023-08-16:
# 🚀 Feature request
Any navigation app support such as google map

## 📝 Description
Phone is minimal from distraction and you are successful with that but it is also minimal in feature. GPS and turn by tune navigation is essential now a days. 

I am aware that it might now be possible as there is no GPS hardware in Phone. However this might be essential to many people. 

## 📝 Describe the solution you'd like

At least google map support 

## 📝 Describe alternatives you've considered

there are other navigation library are there but I feel google are the best in market.

Lefucjusz, 2023-08-21:
Hi @alok-dks, thanks for creating this feature request! Unfortunately, as you have noticed, Pure's hardware does not have onboard GPS module and that makes implementing such feature impossible.

closed 2023-08-21 by Lefucjusz

alok-dks, 2023-08-28:
@Lefucjusz Is there any future plan to integrate onboard GPS module?