mudita/MuditaOS #4857: Bedside light intensity

demoive, 2024-08-05:
# 🚀 Feature request

## 📝 Description

I keep _Frontlight_ intensity to a **minimum** for when I need to see the screen at night. But when I use the Bedside Lamp ([v1.8.0](https://support.mudita.com/en/support/solutions/articles/77000536224-mudita-harmony-1-8-0)) shortcut, I need it on **maximum** intensity to light my environment, read a book, etc.

I see these functionalities serving two different (and opposite) purposes:

1. **Frontlight intensity (setting):** Helps users _see the screen_ – eyes are directly looking at the screen, especially in situations where eyes are acclimating to the dark.
2. **Bedside Lamp (shortcut):** Helps users _light up a dark environment_ – eyes are explicitly not looking at the screen.

## 📝 Describe the solution you'd like

Make the Beside Light always operates in max light intensity.

## 📝 Describe alternatives you've considered

Alternatively, add a separate intensity option to the Frontlight setting (e.g. `Bedside lamp intensity`) which defaults to max intensity.

Lefucjusz, 2024-08-07:
Hi @demoive, I've passed your idea to the team and we'll discuss its feasibility. I'll come back to you as soon as I have more information. Thanks for your suggestion!

Lefucjusz, 2024-08-12:
Hi @demoive, I'm returning to you with great news - the team has decided to implement this feature! It will be released in one of the next OS updates, I'll keep you informed. [I've already implemented a proof-of-concept](https://github.com/mudita/MuditaOS/commits/customizable_bedside_lamp/) if you'd like to look at the code :wink: 

Once again thanks for your suggestion!

Lefucjusz, 2024-10-21:
Hi @demoive, Harmony 2.9.0, containing customizable bedside lamp feature you asked for, has just been released! Please update your Harmony via Mudita Center, you'll find this feature under Menu -> Settings -> Bedside Lamp.

closed 2024-11-04 by Lefucjusz