mudita/MuditaOS #4858: Bedtime chime volume setting

demoive, 2024-08-05:
# 🚀 Feature request

## 📝 Description

The first day I set up a bedtime chime, it took me completely by surprise:

- it almost woke up my kids
- it scared the shit out of me (I didn't recognise the sound and couldn't tell where it was coming from

I'm currently not using this feature due to the above.

## 📝 Describe the solution you'd like

Add a "volume level" option in the settings for Bedtime chime. This has the added benefit to allow the chime to play while changing the volume which helps recognise the expected sound to at bedtime.

Lefucjusz, 2024-08-06:
Hi @demoive, thanks for your suggestion! This option is already available, though - please navigate to Menu -> Settings -> Bedtime -> select bedtime chime you like -> accept it by light click -> you'll be switched to screen where you can set the volume of bedtime notification as per your preference.

demoive, 2024-08-07:
You're right! Not sure how I missed that. Thanks, closing issue...

closed 2024-08-07 by demoive