mudita/MuditaOS #4863: Home screen becoming blurry over time

demoive, 2024-08-07:
# 🐛 Bug report

## 📝 Description

Since the clock face does not regularly perform a "hard refresh", over time the [e-ink "ghosting" effect](https://electronics.stackexchange.com/questions/20276/why-does-flashing-prevent-ghosting-on-e-ink-displays) causes the display to become "dirty" and blurry. The effect worsens/accumulates over time and is more prominent with the backlight on.

## 👍 Expected behavior

I expect* the clock face to be a sharp and clean when I look at it.

*We love Mudita for the attention to detail, simplicity and "clean" experience/interface :-)

## 👎 Current behavior

The primary numbers on the clock face becomes increasingly dirty/blurry over time.

## 🔬 Minimal reproduction

- With the device on (any) home screen...
- Let the numbers on the clock change over a period of a few minutes

You'll notice that as the numbers have changed over time, the (natural) ghosting effect of the e-ink display accumulates and begins to make the screen dirty, giving the impression it is "blurry". The effect is more noticeable if you turn the backlight on and with numbers that have different width – like 8 ↔ 1 – which causes a more drastic horizontal shifting since text alignment is centred.

## 🌍 Your operating system and MuditaOS version

- Harmony 2
- OS v2.8.0

## 🤔 Anything else relevant?

Knowing that this is just an unfortunate side effect of e-ink displays, a possible fix (as a workaround) is to trigger a "hard screen" refresh on a regular interval (e.g. top of every hour). This would clear out the ghosting effect in the same way that returning to the home screen from the menu currently does.

Lefucjusz, 2024-08-08:
Hi @demoive! As you already said, the effect that you observe is caused by the very nature of e-ink displays. The solution that you have proposed (performing deep refresh on a regular interval) is already present in the OS: https://github.com/mudita/MuditaOS/blob/08651cef20a99e07a53658c3547c8a8fd30ff4f5/products/BellHybrid/apps/application-bell-main/presenters/HomeScreenPresenter.cpp#L215

The value of 1 deep refresh every 30 minutes was empirically chosen to be the best compromise between "blurriness" and display blinking that some users find annoying.

Lefucjusz, 2024-11-04:
Closing due to lack of activity.

closed 2024-11-04 by Lefucjusz