mudita/MuditaOS #4878: On clock face, see bedtime and cancel bedtime for one night

nclm, 2024-09-15:
# 🚀 Feature request

## 📝 Description

Right now, bedtime is almost just like another alarm. It does its small ring at the time you set it.

However, if you’re already in bed early and sleeping before the bedtime alarm, it might just wake you up, which is the opposite of what you want. I would use bedtime on days where I don’t notice the late time, but I have it off because I also go to bed earlier other times and don’t want to be waken up.

## 📝 Describe the solution you'd like

Have the fact that bedtime is turned on **shown on the clockface**, just like having alarm on, so you cannot forget you have it on. A little moon icon with its time, just as alarm is a little bell icon with its time, that you can glimpse when going to bed if you go earlier.

Have a way to deactivate it with one button action straight from the clockface, to **cancel it just for tonight**. So you can press it just before going to sleep. It could be the back button for instance, which is unused on clockface.

## 📝 Describe alternatives you've considered

There could also be something even more visible in the few hours that precede bed time. Just to be sure you don’t miss it and cancel it before you go to bed, but can also be a reminder to go to bed if you glance at the clock. It could be a small text just like when turning an alarm on: “Bedtime will ring in 1 h & 14 min”

If there is any light censor in the clock, there could be also something that prevents it to ring if it’s dark.

ps. It could also be nice to have more options for bedtime, i.e. sound and volume, but that’s another issue.

Lefucjusz, 2024-09-16:
Hi @nclm, thanks for your suggestion! We will consider implementing your idea in the future. We'll keep you informed.

marten-seemann, 2026-09-07:
How about just disabling the bedtime chime if the alarm is already set? No extra controls needed, (and without any knowledge of the code base, so I could be wrong) should be pretty straightforward to implement?

nclm, 2026-09-07:
> How about just disabling the bedtime chime if the alarm is already set? No extra controls needed, (and without any knowledge of the code base, so I could be wrong) should be pretty straightforward to implement?

That’s not solving the original issue? If I have an alarm set at 7:00 in the morning and a bedtime set at 23:00, I still want the bedtime to ring at 23:00 if I’m still up by that time, but be able to cancel it for tonight if I got to bed at 22:00.

marten-seemann, 2026-09-07:
You might be using your clock differently than me, but I don’t arm it the morning before, but just before I go to bed.

nclm, 2026-09-07:
> You might be using your clock differently than me, but I don’t arm it the morning before, but just before I go to bed.

I see, makes sense. I don’t disarm it when it rings and keep it on for the next day, only disabling it on weekends.